kitspace/kitspace #35: Title line spacing issue in mobile view

kasbah, 2016-05-05:
[image]

slifin, 2016-05-06:
I'm having trouble replicating this
what device/browser did you use to replicate this? 

Mobile Firefox (Nexus 5)
[image]
Mobile Chrome (Nexus 5)
[image]
iphone 5 (emulated) 
[image]
other emulated devices yield similar results

kasbah, 2016-05-06:
Firefox responsive design mode with a width less than 480 or so. 
[image]

closed 2016-05-10 by kasbah