kitspace/kitspace #64: Graphics issues in Opera

kasbah, 2016-05-15:
https://saucelabs.com/beta/tests/34db96409a694e0086bb51d8661da393

Alignment in titlebar is off

[image]

SVGs don't appear on main page and size of Mouser icon is wrong.

[image]

slifin, 2016-05-18:
Are you sure you want me to try and get Opera 12/11 rendering correctly? 
For reference the current version of Opera is 37 and runs on a different rendering engine

kasbah, 2016-05-18:
Ah, I wasn't aware. Should briefly test a recent Opera though.

slifin, 2016-05-19:
Yeah I took a look on OSX yesterday everything seems good there on Opera, I assume this is because Chrome and Opera share blink as a rendering engine

Here is a screenshot of windows: 

[image]

kasbah, 2016-05-19:
cool

closed 2016-05-19 by kasbah