kitspace/kitspace #72: Get rid of absolute positioning of title?

kasbah, 2016-05-18:
Is the absolute positioning still necessary? I was trying to make some spacing adjustments of the other elements and it's making things difficult.

slifin, 2016-05-19:
The absolute positioning is to ensure this scenario doesn't happen: http://jsfiddle.net/cvz5vefc/
If we can drop that centering as a requirement then we can put it back into the flex container or there might be another solution using percentage widths, it just means we would be relying on the break point to collapse the menu into vertical rather than flex which can wrap automatically when space runs out

kasbah, 2016-05-19:
Can we not just make the logo and the button bit the same width?

slifin, 2016-05-19:
what do you think about this solution? 
https://jsfiddle.net/4w1brw1k/

kasbah, 2016-05-19:
Looks good, yeah.

kasbah, 2020-02-16:
I think this was solved with the redesign with #185

closed 2020-02-16 by kasbah