kitspace/kitspace #77: Scrape BOM prices from distributors

JarrettR, 2016-05-25:
...And display them in the project. This might have to be done regularly to account for market prices.

kasbah, 2016-05-25:
Thanks for the idea Jarret! At some point will be moving over to having a proper server side for Kitnic and this sort of feature will be easier then.

Though we could add pricing information statically, at build time, in the meantime and use https://freegeoip.kitnic.it to change the content based on location. The site gets re-built pretty regularly at the moment so getting out of date wouldn't be too much of an issue. 

Overall the build times are going to get ridiculous though as more projects are added so that server side change needs to happen anyway.

kasbah, 2020-02-16:
Closing this as this type of thing is planned for https://github.com/kitspace/partinfo

closed 2020-02-16 by kasbah