kitspace/kitspace #88: Adding parts to cart doesn't work for Mouser/Digikey

descampsa, 2016-10-02:
You 'add parts to cart' functionnality seems to have problems with Mouser and Digikey. 
For both suppliers, it never complete and seems to give up after some time.
This happen with the website and the chrome extension.
For Digikey, it works partially (for example, with this [board](https://kitnic.it/boards/github.com/kasbah/push-on-hold-off/), i end up with 7 items in my cart, instead of 11, and the webpage never open, i have to go to digikey.com by myself.
For Mouser, i was not able to make it work at all.

kasbah, 2016-10-02:
Thanks for reporting! I had a look and quickly released a fix for Digikey with version 1.1.7. 

For Mouser I am not sure what is going on, it works for me. Could you make sure the correct country is set in the options and try again? Do you have any other extensions installed? Could you try and disable some and see if they are interfering?

descampsa, 2016-10-02:
I confirm that 1.1.7 fixes the issue.

For Mouser, chrome reports an error in your code that might be the cause of the issue.
Here is the stack trace :

Uncaught TypeError: errors.forEach is not a function
(anonymous function) @ mouser.js:122
network_callback @ http.js:29
xhr.onreadystatechange @ http.js:90

My version of chrome is:
Version 47.0.2526.80 Built on 8.2, running on Debian 8.5 (64-bit)

Let me know if you need more info.

kasbah, 2016-10-02:
I can't reproduce this on Chrome 52 but I can see why there could be a problem with that line and I have changed it.  Could you download this, unzip it and load it as an unpacked extension and see if it fixes it?

[1clickBOM-v1.1.7.1-chrome.zip](https://github.com/monostable/kitnic/files/505166/1clickBOM-v1.1.7.1-chrome.zip)

Need to remember to check older versions of browsers.

descampsa, 2016-10-02:
Yes, the problem is fixed with that new version, perfect!

kasbah, 2016-10-02:
Great, I have released version 1.1.8 with this fix. Thanks very much for reporting and testing!

closed 2016-10-02 by kasbah