kitspace/kitspace #149: Accomodate multiple kitspace projects in one repo

patcon, 2018-10-30:
For those who've either go a few in one, or are supporting multiple versions (just different folders, in this case), and want to keep project in one authoritative git history.

Our project has multiple versions that we might like to help people order more of (since they might have 100 existing badges to maintain compatibility with) and also a badge programmer:
https://github.com/HumanDynamics/openbadge/tree/master/eagle/nRF_badge

kasbah, 2018-10-30:
The way I would handle multiple versions in my own projects is through git tags. I am planning some UI for this and navigating the git history in general in the future.

Multiple projects per repo would still be useful though. I think in the kitspace.yaml it could be handled by allowing something like:

```yaml
multi:
    project_1:
         gerbers: project_1/gerbers
    project_2:
        gerbers: project_2/gerbers
```

Jan--Henrik, 2019-02-12:
I also need this feature to add my projects to Kitspace.org, accordingly +1 :)

kasbah, 2019-08-07:
This was fixed with #180 !

closed 2019-08-07 by kasbah