kitspace/kitspace #159: Improve use of space on initial page

kasbah, 2019-03-27:
I limited the number of projects per row on the main overview to three as I think it makes it easy to scan them. There is however a lot of space left for other things. 

[image]

I already started re-designing the top bar for #112 

[image]

What could we do with the initial page in general?

jOnoNe, 2019-04-05:
[image]

jOnoNe, 2019-04-17:
https://projects.invisionapp.com/share/ZVRJZFFSMXY#/screens/358251783_Front_Page

kasbah, 2019-04-18:
Oh, interesting, it's a bit of a departure from what I had envisioned. You are introducing a whole new design system?

The design system for Kitspace was of course never formally defined, with me just hacking away at it. But I did start relying heavily on the system built into [Semantic-UI](https://semantic-ui.com/). I still really like this system:

1. It riffs of established design systems, especially GitHub's, making it easier for a new user to navigate
2. It doesn't grab attention, except for actions like "submit a project" and the focus stays on the (colorful) projects themselves
3. Using it saves a ton of time as a lot of decisions are made for your

So: I don't like the use of color and PCB aesthetic as it takes the focus away from the projects. I do like the arrangement of the topics and the really short description and the links.

One other thing remains: it doesn't actually solve the problem I opened this issue for (I will rename the issue though to be more general though). The use of space on either side of the projects list. The fundamental conflict remains in your new concept: you also have 3 projects per row so on larger screens the sides are empty. 

I think having 3 projects per row makes sense as it's less easy to scan the list with more. Thinking on it now it may be possible to scale the size of the project cards but not without incurring some cost in terms of image download size. 

Would be cool if we can arrange to meet up in person or on video chat soon to go through everything. Can you ping me by email or chat when you could do that?

jOnoNe, 2019-04-18:
Semantic UI looks good! and I agree it makes some good decisions without us
having to. I was just playing around with a concept for the styling - you
should address styling at some point, but it doesn't have to be now. If you
want to attract maker-types and businesses, you will benefit from a custom
graphic language.

I agree 3 rows is plenty of info. You could enlarge the grid on larger
screens, but it is good practice to always have a max width. White space is
an important part of a design. If you wanted to do this there are
techniques such as loading a lower quality image first while waiting for
the higher quality image to load. But I don't think it is really a
significant problem at the moment and I would leave it how it is.

What the prototype does address is more efficient use of header space,
concisely explaining what the user can expect the site to do and suggests a
way filters could work. Basic filtering like this would be the easiest way
to improve the browsing project experience.

jOnoNe, 2019-04-23:
Todo: Add Riot.im, Twitter. Remove Reddit and Hackaday. Add 'login'. Remove styling and use default Semantic UI

jOnoNe, 2019-08-02:
Wireframe for the top nav: 
[image]

@kasbah

PR #185 (2019-08-04): Improve front page and title bar
Adds pages bout the BOM Builder and 1-click BOM, shortens the intro and makes the nav-bar nicer. 

closes #159
closed 2019-08-04 by kasbah