kitspace/kitspace #160: ./ninja errors

evanyang1, 2019-03-30:
When I type in 
$ ./ninja
I get this error
[image]

kasbah, 2019-03-30:
I think I gave you the wrong command to set your path in the chat. Use:

```
export PATH="$(pwd)/node_modules/.bin:$PATH"
```
(notice the quote marks)

Another issue may be that you do need to use node v6. I would recommend you set up [nvm](https://github.com/creationix/nvm#install--update-script) and switch to using v6 for now.

evanyang1, 2019-03-30:
[image]

kasbah, 2019-03-30:
Does `node -v` return 6.x?

evanyang1, 2019-03-30:
Oh I have 10.x

kasbah, 2019-03-30:
See my initial reply about using node 6.

evanyang1, 2019-03-30:
[image]
 After `npm install` again. Smaller error log this time.

kasbah, 2019-03-30:
Google that error along with linux subsystem for windows and see what you get. By the way if you pasted text into the issue instead of images it would be easier for me to google it ;).

evanyang1, 2019-03-30:
./ninja
[1/32] node -- tasks/page/makePage.js src/page/page.html -- dev build/boards/github.com/kitspace-forks/Bus_Pirate/index.html
FAILED: build/boards/github.com/kitspace-forks/Bus_Pirate/index.html
browserify --list tasks/page/makePage.js > build/boards/github.com/kitspace-forks/Bus_Pirate/index.html.d         && if [ '' != '' ]; then browserify --extension='.jsx' --transform [ babelify     --presets [ react es2015 ] ] --debug --list  >> build/boards/github.com/kitspace-forks/Bus_Pirate/index.html.d; fi         && ./scripts/depfileify build/boards/github.com/kitspace-forks/Bus_Pirate/index.html build/boards/github.com/kitspace-forks/Bus_Pirate/index.html.d         && node -- tasks/page/makePage.js src/page/page.html -- dev build/boards/github.com/kitspace-forks/Bus_Pirate/index.html
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: ENOENT: no such file or directory, lstat '/mnt/c/Users/Evan Yang/Desktop/kitspace/node_modules/fbjs/lib/warning.js'
    at Error (native)
[6/32] browserify -x 1-click-bom -x babel-polyfill -x browser-version -x gerber-to-sv...ber -x xml-element-string build/.temp/submit/render_submit.jsx -o build/submit/app.j
ninja: build stopped: subcommand failed.

kasbah, 2019-03-30:
Looks like this is solved for now. Some notes on running the build on Windows sub-system for Linux: 

- Use node 6
- Get ninja and inkscape from apt
- use `/home/` as other directories are not the right file system

closed 2019-03-30 by kasbah