kitspace/kitspace #168: Dev Environment Setup - Error: EISDIR: illegal operation on a directory, read

SDBowen, 2019-04-29:
Hello :wave: 

Ran into the error below trying to get a dev server up:

[image]

I am on Ubuntu and used nvm to run node v6.17.1.

Error text:

[9/20] node -- tasks/page/processGerbers.js boards/github.com/kitspace-forks/arduino-uno boards/github.c...no/images/top-meta.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-with-background.png
FAILED: build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg 
browserify --list tasks/page/processGerbers.js > build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg.d         && if [ '' != '' ]; then browserify --extension='.jsx' --transform [ babelify     --presets [ react es2015 ] ] --debug --list  >> build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg.d; fi         && ./scripts/depfileify build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg.d         && node -- tasks/page/processGerbers.js boards/github.com/kitspace-forks/arduino-uno boards/github.com/kitspace-forks/arduino-uno/1-click-bom.tsv boards/github.com/kitspace-forks/arduino-uno/bom boards/github.com/kitspace-forks/arduino-uno/bom/20150523_untitled_bom.csv boards/github.com/kitspace-forks/arduino-uno/bom/arduino_Uno_Rev3-02-TH.txt boards/github.com/kitspace-forks/arduino-uno/cad boards/github.com/kitspace-forks/arduino-uno/cad/arduino_Uno_Rev3-02-TH.brd boards/github.com/kitspace-forks/arduino-uno/cad/arduino_Uno_Rev3-02-TH.sch boards/github.com/kitspace-forks/arduino-uno/gerbers boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.cmp boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.drd boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.dri boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.gko boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.gpi boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.plc boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.sol boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.stc boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.sts boards/github.com/kitspace-forks/arduino-uno/kitnic.yaml boards/github.com/kitspace-forks/arduino-uno/README.md -- dev build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg build/boards/github.com/kitspace-forks/arduino-uno/images/bottom.svg build/boards/github.com/kitspace-forks/arduino-uno/arduino-uno-5f10c3e-gerbers.zip build/.temp/boards/github.com/kitspace-forks/arduino-uno/zip-info.json build/.temp/boards/github.com/kitspace-forks/arduino-uno/unoptimized-top.svg build/boards/github.com/kitspace-forks/arduino-uno/images/top.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-large.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-meta.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-with-background.png
Could not process gerbers for boards/github.com/kitspace-forks/arduino-uno
{ Error: EISDIR: illegal operation on a directory, read
    at Error (native)
    at Object.fs.readSync (fs.js:733:19)
    at tryReadSync (fs.js:488:20)
    at Object.fs.readFileSync (fs.js:536:19)
    at Object.<anonymous> (/home/steven/kitspace/tasks/page/processGerbers.js:85:17)
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3) errno: -21, code: 'EISDIR', syscall: 'read' }
[10/20] node -- tasks/page/processGerbers.js boards/github.com/kitspace-forks/Bus_Pirate boards/github.c...ate/images/top-meta.png build/boards/github.com/kitspace-forks/Bus_Pirate/images/top-with-background.png
FAILED: build/boards/github.com/kitspace-forks/Bus_Pirate/images/top.svg 
browserify --list tasks/page/processGerbers.js > build/boards/github.com/kitspace-forks/Bus_Pirate/images/top.svg.d         && if [ '' != '' ]; then browserify --extension='.jsx' --transform [ babelify     --presets [ react es2015 ] ] --debug --list  >> build/boards/github.com/kitspace-forks/Bus_Pirate/images/top.svg.d; fi         && ./scripts/depfileify build/boards/github.com/kitspace-forks/Bus_Pirate/images/top.svg build/boards/github.com/kitspace-forks/Bus_Pirate/images/top.svg.d         && node -- tasks/page/processGerbers.js boards/github.com/kitspace-forks/Bus_Pirate boards/github.com/kitspace-forks/Bus_Pirate/1-click-bom.tsv boards/github.com/kitspace-forks/Bus_Pirate/cad boards/github.com/kitspace-forks/Bus_Pirate/cad/BusPirate-v3.6a-SSOP.brd boards/github.com/kitspace-forks/Bus_Pirate/cad/BusPirate-v3.6a-SSOP.sch boards/github.com/kitspace-forks/Bus_Pirate/gerbers boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.cmp boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.drd boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.dri boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.gko boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.gpi boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.plc boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.pls boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.sol boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.stc boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.sts boards/github.com/kitspace-forks/Bus_Pirate/kitnic.yaml -- dev build/boards/github.com/kitspace-forks/Bus_Pirate/images/top.svg build/boards/github.com/kitspace-forks/Bus_Pirate/images/bottom.svg build/boards/github.com/kitspace-forks/Bus_Pirate/Bus_Pirate-0ca376a-gerbers.zip build/.temp/boards/github.com/kitspace-forks/Bus_Pirate/zip-info.json build/.temp/boards/github.com/kitspace-forks/Bus_Pirate/unoptimized-top.svg build/boards/github.com/kitspace-forks/Bus_Pirate/images/top.png build/boards/github.com/kitspace-forks/Bus_Pirate/images/top-large.png build/boards/github.com/kitspace-forks/Bus_Pirate/images/top-meta.png build/boards/github.com/kitspace-forks/Bus_Pirate/images/top-with-background.png
Could not process gerbers for boards/github.com/kitspace-forks/Bus_Pirate
{ Error: EISDIR: illegal operation on a directory, read
    at Error (native)
    at Object.fs.readSync (fs.js:733:19)
    at tryReadSync (fs.js:488:20)
    at Object.fs.readFileSync (fs.js:536:19)
    at Object.<anonymous> (/home/steven/kitspace/tasks/page/processGerbers.js:85:17)
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3) errno: -21, code: 'EISDIR', syscall: 'read' }
[13/20] node -- tasks/page/processGerbers.js boards/github.com/mattvenn/esp8266-breakout boards/github.c...out/images/top-meta.png build/boards/github.com/mattvenn/esp8266-breakout/images/top-with-background.png
FAILED: build/boards/github.com/mattvenn/esp8266-breakout/images/top.svg 
browserify --list tasks/page/processGerbers.js > build/boards/github.com/mattvenn/esp8266-breakout/images/top.svg.d         && if [ '' != '' ]; then browserify --extension='.jsx' --transform [ babelify     --presets [ react es2015 ] ] --debug --list  >> build/boards/github.com/mattvenn/esp8266-breakout/images/top.svg.d; fi         && ./scripts/depfileify build/boards/github.com/mattvenn/esp8266-breakout/images/top.svg build/boards/github.com/mattvenn/esp8266-breakout/images/top.svg.d         && node -- tasks/page/processGerbers.js boards/github.com/mattvenn/esp8266-breakout boards/github.com/mattvenn/esp8266-breakout/1-click-bom.tsv boards/github.com/mattvenn/esp8266-breakout/esp8266-12-breakout.kicad_pcb boards/github.com/mattvenn/esp8266-breakout/esp8266-12-breakout.net boards/github.com/mattvenn/esp8266-breakout/esp8266-12-breakout.pro boards/github.com/mattvenn/esp8266-breakout/esp8266-12-breakout.sch boards/github.com/mattvenn/esp8266-breakout/gerbers boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-B.Cu.gbr boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-B.Mask.gbr boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-B.SilkS.gbr boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-1 [... cut ...]
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3) errno: -21, code: 'EISDIR', syscall: 'read' }
[14/20] node -- tasks/page/processBOM.js build/.temp/boards.json boards/github.com/kitspace-forks/arduin.../kitspace-forks/arduino-uno/info.json build/boards/github.com/kitspace-forks/arduino-uno/1-click-BOM.tsv
ninja: build stopped: subcommand failed.
BUILD FAILED

kasbah, 2019-04-29:
Hi Steven, thanks for reporting. I edited your comment to format the error text in a more readable way.

As for your issue it looks like it's getting a directory in the list of gerber files and trying to read it like a file. Not sure why though and I can't reproduce it. I have seen this when people use node 8 so the only thing I can think of is that you are not actually running node 6 when you think you are. Please double check that.

If you are really sure you are running on 6, I also just pushed a branch called `debug-read-on-dir` which `console.log`'s the gerber list. Could you pull that and paste the output?

SDBowen, 2019-04-29:
Appreciate the help.

I am fairly confident I am running node 6:

[image]

Here is the output from the debug branch:

~~~
generated ./build.ninja with 'dev' config
ninja
[172/216] node-sass --sourcemap=none --load-path src/ src/page/page.scss build/.temp/page.css
Rendering Complete, saving .css file...
Wrote CSS to /home/steven/kitspace/build/.temp/page.css
[173/216] node-sass --sourcemap=none --load-path src/ src/index/index.scss build/.temp/index.css
Rendering Complete, saving .css file...
Wrote CSS to /home/steven/kitspace/build/.temp/index.css
[174/216] node-sass --sourcemap=none --load-path src/ src/submit/submit.scss build/.temp/submit.css
Rendering Complete, saving .css file...
Wrote CSS to /home/steven/kitspace/build/.temp/submit.css
[189/216] node -- tasks/page/processGerbers.js boar...ce-forks/arduino-uno/images/top-with-background.png
FAILED: build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg 
browserify --list tasks/page/processGerbers.js > build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg.d         && if [ '' != '' ]; then browserify --extension='.jsx' --transform [ babelify     --presets [ react es2015 ] ] --debug --list  >> build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg.d; fi         && ./scripts/depfileify build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg.d         && node -- tasks/page/processGerbers.js boards/github.com/kitspace-forks/arduino-uno boards/github.com/kitspace-forks/arduino-uno/1-click-bom.tsv boards/github.com/kitspace-forks/arduino-uno/bom boards/github.com/kitspace-forks/arduino-uno/bom/20150523_untitled_bom.csv boards/github.com/kitspace-forks/arduino-uno/bom/arduino_Uno_Rev3-02-TH.txt boards/github.com/kitspace-forks/arduino-uno/cad boards/github.com/kitspace-forks/arduino-uno/cad/arduino_Uno_Rev3-02-TH.brd boards/github.com/kitspace-forks/arduino-uno/cad/arduino_Uno_Rev3-02-TH.sch boards/github.com/kitspace-forks/arduino-uno/gerbers boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.cmp boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.drd boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.dri boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.gko boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.gpi boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.plc boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.sol boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.stc boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.sts boards/github.com/kitspace-forks/arduino-uno/kitnic.yaml boards/github.com/kitspace-forks/arduino-uno/README.md -- dev build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg build/boards/github.com/kitspace-forks/arduino-uno/images/bottom.svg build/boards/github.com/kitspace-forks/arduino-uno/arduino-uno-5f10c3e-gerbers.zip build/.temp/boards/github.com/kitspace-forks/arduino-uno/zip-info.json build/.temp/boards/github.com/kitspace-forks/arduino-uno/unoptimized-top.svg build/boards/github.com/kitspace-forks/arduino-uno/images/top.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-large.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-meta.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-with-background.png
{ gerbers: 
   [ 'boards/github.com/kitspace-forks/arduino-uno/1-click-bom.tsv',
     'boards/github.com/kitspace-forks/arduino-uno/bom',
     'boards/github.com/kitspace-forks/arduino-uno/bom/20150523_untitled_bom.csv',
     'boards/github.com/kitspace-forks/arduino-uno/bom/arduino_Uno_Rev3-02-TH.txt',
     'boards/github.com/kitspace-forks/arduino-uno/cad',
     'boards/github.com/kitspace-forks/arduino-uno/cad/arduino_Uno_Rev3-02-TH.brd',
     'boards/github.com/kitspace-forks/arduino-uno/cad/arduino_Uno_Rev3-02-TH.sch',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.cmp',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.drd',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.dri',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.gko',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.gpi',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.plc',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.sol',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.stc',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.sts',
     'boards/github.com/kitspace-forks/arduino-uno/kitnic.yaml',
     'boards/github.com/kitspace-forks/arduino-uno/README.md' ] }
Could not process gerbers for boards/github.com/kitspace-forks/arduino-uno
{ Error: EISDIR: illegal operation on a directory, read
    at Error (native)
    at Object.fs.readSync (fs.js:733:19)
    at tryReadSync (fs.js:488:20)
    at Object.fs.readFileSync (fs.js:536:19)
    at Object.<anonymous> (/home/steven/kitspace/tasks/page/processGerbers.js:86:17)
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
[... cut ...]
    at Object.<anonymous> (/home/steven/kitspace/tasks/page/processGerbers.js:86:17)
    at Module._compile (module.js:577:32)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3) errno: -21, code: 'EISDIR', syscall: 'read' }
[194/216] node -- tasks/page/makeReadme.js build/.t...ds/github.com/kitspace-forks/arduino-uno/readme.jsx
ninja: build stopped: subcommand failed.
BUILD FAILED
watching for file changes in src/ tasks/ boards/
~~~

kasbah, 2019-04-29:
Right, looks like there are actually directories in that list of gerbers, though I don't quite know why. They should have been filtered out by the `gerberFiles` function. 

I pushed another commit that filters out directories earlier on, if you can pull the `debug-read-on-dir` and test, that would help.

I suspect it will get us past this error but still have too many files that it thinks are gerbers but are not.

As I am writing this I just thought of something else: could you use https://yarnpkg.com instead of `npm`? We have a yarn lock file that locks down dependencies to an exact version but don't have the same for `npm`.

SDBowen, 2019-04-29:
Success! :confetti_ball:

I cloned the repo again and started fresh using Yarn.

~~~
yarn run v1.15.2
$ ./scripts/get_boards && npm run serve & npm run watch
git clone --depth=1 https://github.com/mattvenn/esp8266-breakout boards/github.com/mattvenn/esp8266-breakout && cd boards/github.com/mattvenn/esp8266-breakout && git checkout fcb3107c66670035c7cb330f5d98eddde5d17f56 || git fetch --unshallow && git checkout fcb3107c66670035c7cb330f5d98eddde5d17f56
git clone --depth=1 https://github.com/kitspace-forks/arduino-uno boards/github.com/kitspace-forks/arduino-uno && cd boards/github.com/kitspace-forks/arduino-uno && git checkout 5f10c3e4d9709344153aecf65faaef492ddcb0d1 || git fetch --unshallow && git checkout 5f10c3e4d9709344153aecf65faaef492ddcb0d1
git clone --depth=1 https://github.com/kitspace-forks/Bus_Pirate boards/github.com/kitspace-forks/Bus_Pirate && cd boards/github.com/kitspace-forks/Bus_Pirate && git checkout 0ca376a0eedd116e3ffdbd5dfd5398a67fdba708 || git fetch --unshallow && git checkout 0ca376a0eedd116e3ffdbd5dfd5398a67fdba708

> kitspace@1.0.0 watch /home/steven/kitspace
> ./scripts/watch

> kitspace@1.0.0 serve /home/steven/kitspace
> http-server build/

Starting up http-server, serving build/
Available on:
  http://127.0.0.1:8082
  http://192.168.0.36:8082
Hit CTRL-C to stop the server
generated ./build.ninja with 'dev' config
ninja
[172/206] node-sass --sourcemap=none --load-path src/ src/submit/submit.scss build/.temp/submit.css
Rendering Complete, saving .css file...
Wrote CSS to /home/steven/kitspace/build/.temp/submit.css
[173/206] node-sass --sourcemap=none --load-path src/ src/page/page.scss build/.temp/page.css
Rendering Complete, saving .css file...
Wrote CSS to /home/steven/kitspace/build/.temp/page.css
[174/206] node-sass --sourcemap=none --load-path src/ src/index/index.scss build/.temp/index.css
Rendering Complete, saving .css file...
Wrote CSS to /home/steven/kitspace/build/.temp/index.css
[190/206] node -- tasks/page/processGerbers.js boards/github.com/kitspace-forks/arduino-uno boards/githu...no/images/top-meta.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-with-background.png
{ gerbers: 
   [ 'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.cmp',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.drd',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.gko',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.plc',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.sol',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.stc',
     'boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.sts' ] }
[199/206] node -- tasks/page/processGerbers.js boards/github.com/mattvenn/esp8266-breakout boards/github...out/images/top-meta.png build/boards/github.com/mattvenn/esp8266-breakout/images/top-with-background.png
{ gerbers: 
   [ 'boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-B.Cu.gbr',
     'boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-B.Mask.gbr',
     'boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-B.SilkS.gbr',
     'boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-Edge.Cuts.gbr',
     'boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-F.Cu.gbr',
     'boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-F.Mask.gbr',
     'boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-F.SilkS.gbr',
     'boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-NPTH.drl',
     'boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout.drl' ] }
[205/206] node -- tasks/page/processGerbers.js boards/github.com/kitspace-forks/Bus_Pirate boards/github...ate/images/top-meta.png build/boards/github.com/kitspace-forks/Bus_Pirate/images/top-with-background.png
{ gerbers: 
   [ 'boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.cmp',
     'boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.drd',
     'boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.gko',
     'boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.plc',
     'boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.pls',
     'boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.sol',
     'boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.stc',
     'boards/github.com/kitspace-forks/Bus_Pirate/gerbers/BusPirate-v3.6a-SSOP.sts' ] }
[206/206] browserify -x 1-click-bom -x babel-polyfill -x browser-version -x gerber-to-svg -x github-url-...tspace-forks/Bus_Pirate/page/render_page.jsx -o build/boards/github.com/kitspace-forks/Bus_Pirate/app.js
* build succeeded *
watching for file changes in src/ tasks/ boards/
~~~

kasbah, 2019-04-29:
Great! I'll keep this open till me or someone else changes the docs in README. 

Can I ask what are you planning to do with the Kitspace site?

SDBowen, 2019-04-29:
I had not thought that far ahead yet. :smile:

I saw the project on [https://findcollabs.com](url) and thought I would take and look and see where I can help out. Is there an open issue you would suggest as a good starting point?

kasbah, 2019-04-30:
That's great. You could fix the docs for npm vs yarn as a first issue :)

After that, I have been waiting on @evanyang1 to do #155 for quite a while, maybe it's time for someone else to pick that up and @evanyang1 can pick up something else when they have time again.

kasbah, 2019-04-30:
Or as a smaller issue to start of with, I just opened #170

PR #171 (2019-05-01): Update README to recommend 'yarn install'
Update to the README to recommend `yarn install` be used. A link to  [ https://yarnpkg.com]( https://yarnpkg.com) was also added under the Requirements section.

This ensures the yarn.lock file is used and correct versions of dependencies are installed.

closes: #168
closed 2019-05-01 by kasbah