kitspace/kitspace #173: Build issue

evanyang1, 2019-05-01:
Used `yarn && yarn start` and got this error: (ran $ export PATH="$(pwd)/node_modules/.bin:$PATH" )

yarn install v1.15.2
[1/4] Resolving packages...
success Already up-to-date.
Done in 0.84s.
yarn run v1.15.2
$ ./scripts/get_boards && npm run serve & npm run watch
cd boards/github.com/mattvenn/esp8266-breakout && git checkout fcb3107c66670035c7cb330f5d98eddde5d17f56 || git fetch --unshallow && git checkout fcb3107c66670035c7cb330f5d98eddde5d17f56 || git pull origin master && git checkout fcb3107c66670035c7cb330f5d98eddde5d17f56
cd boards/github.com/kitspace-forks/Bus_Pirate && git checkout 0ca376a0eedd116e3ffdbd5dfd5398a67fdba708 || git fetch --unshallow && git checkout 0ca376a0eedd116e3ffdbd5dfd5398a67fdba708 || git pull origin master && git checkout 0ca376a0eedd116e3ffdbd5dfd5398a67fdba708
cd boards/github.com/kitspace-forks/arduino-uno && git checkout 5f10c3e4d9709344153aecf65faaef492ddcb0d1 || git fetch --unshallow && git checkout 5f10c3e4d9709344153aecf65faaef492ddcb0d1 || git pull origin master && git checkout 5f10c3e4d9709344153aecf65faaef492ddcb0d1
npm WARN lifecycle The node binary used for scripts is /tmp/yarn--1556754049597-0.08400272336458747/node but npm is using /usr/bin/node itself. Use the `--scripts-prepend-node-path` option to include the path for the node binary npm was executed with.

> kitspace@1.0.0 watch /home/evanyang/Desktop/kitspace
> ./scripts/watch

npm WARN lifecycle The node binary used for scripts is /tmp/yarn--1556754049597-0.08400272336458747/node but npm is using /usr/bin/node itself. Use the `--scripts-prepend-node-path` option to include the path for the node binary npm was executed with.

> kitspace@1.0.0 serve /home/evanyang/Desktop/kitspace
> http-server build/

Starting up http-server, serving build/
Available on:
  http://127.0.0.1:8081
  http://192.168.0.24:8081
Hit CTRL-C to stop the server
generated ./build.ninja with 'dev' config
ninja
[1/6] node -- tasks/page/processGerbers.js boards/github.com/mattvenn/esp8266-breakout boards...top-meta.png build/boards/github.com/mattvenn/esp8266-breakout/images/top-with-background.png
FAILED: build/boards/github.com/mattvenn/esp8266-breakout/images/top.svg 
browserify --list tasks/page/processGerbers.js > build/boards/github.com/mattvenn/esp8266-breakout/images/top.svg.d         && if [ '' != '' ]; then browserify --extension='.jsx' --transform [ babelify     --presets [ react es2015 ] ] --debug --list  >> build/boards/github.com/mattvenn/esp8266-breakout/images/top.svg.d; fi         && ./scripts/depfileify build/boards/github.com/mattvenn/esp8266-breakout/images/top.svg build/boards/github.com/mattvenn/esp8266-breakout/images/top.svg.d         && node -- tasks/page/processGerbers.js boards/github.com/mattvenn/esp8266-breakout boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-B.Cu.gbr boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-B.Mask.gbr boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-B.SilkS.gbr boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-Edge.Cuts.gbr boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-F.Cu.gbr boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-F.Mask.gbr boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-F.SilkS.gbr boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout-NPTH.drl boards/github.com/mattvenn/esp8266-breakout/gerbers/esp8266-12-breakout.drl -- dev build/boards/github.com/mattvenn/esp8266-breakout/images/top.svg build/boards/github.com/mattvenn/esp8266-breakout/images/bottom.svg build/boards/github.com/mattvenn/esp8266-breakout/esp8266-breakout-fcb3107-gerbers.zip build/.temp/boards/github.com/mattvenn/esp8266-breakout/zip-info.json build/.temp/boards/github.com/mattvenn/esp8266-breakout/unoptimized-top.svg build/boards/github.com/mattvenn/esp8266-breakout/images/top.png build/boards/github.com/mattvenn/esp8266-breakout/images/top-large.png build/boards/github.com/mattvenn/esp8266-breakout/images/top-meta.png build/boards/github.com/mattvenn/esp8266-breakout/images/top-with-background.png
{ Error: Command failed: inkscape --without-gui 'build/.temp/boards/github.com/mattvenn/esp8266-breakout/unoptimized-top.svg' --export-png='build/boards/github.com/mattvenn/esp8266-breakout/images/top-meta.png' --export-height=400
/bin/sh: 1: inkscape: not found

    at ChildProcess.exithandler (child_process.js:275:12)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Socket.stream.socket.on (internal/child_process.js:346:11)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at Pipe._handle.close [as _onclose] (net.js:557:12)
  killed: false,
  code: 127,
  signal: null,
  cmd: 'inkscape --without-gui \'build/.temp/boards/github.com/mattvenn/esp8266-breakout/unoptimized-top.svg\' --export-png=\'build/boards/github.com/mattvenn/esp8266-breakout/images/top-meta.png\' --export-height=400' }
[2/6] node -- tasks/page/processGerbers.js boards/github.com/kitspace-forks/arduino-uno board...op-meta.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-with-background.png
FAILED: build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg 
browserify --list tasks/page/processGerbers.js > build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg.d         && if [ '' != '' ]; then browserify --extension='.jsx' --transform [ babelify     --presets [ react es2015 ] ] --debug --list  >> build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg.d; fi         && ./scripts/depfileify build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg.d         && node -- tasks/page/processGerbers.js boards/github.com/kitspace-forks/arduino-uno boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.cmp boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.drd boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.gko boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.plc boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.sol boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.stc boards/github.com/kitspace-forks/arduino-uno/gerbers/arduino_Uno_Rev3-02-TH.sts -- dev build/boards/github.com/kitspace-forks/arduino-uno/images/top.svg build/boards/github.com/kitspace-forks/arduino-uno/images/bottom.svg build/boards/github.com/kitspace-forks/arduino-uno/arduino-uno-5f10c3e-gerbers.zip build/.temp/boards/github.com/kitspace-forks/arduino-uno/zip-info.json build/.temp/boards/github.com/kitspace-forks/arduino-uno/unoptimized-top.svg build/boards/github.com/kitspace-forks/arduino-uno/images/top.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-large.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-meta.png build/boards/github.com/kitspace-forks/arduino-uno/images/top-with-background.png
{ Error: Command failed: inkscape --without-gui 'build/.temp/boards/github.com/kitspace-forks/arduino-uno/unoptimized-top.svg' --export-png='build/boards/github.com/kitspace-forks/arduino-uno/images/top-meta.png' --export-height=400
/bin/sh: 1: inkscape: not found

    at ChildProcess.exithandler (child_process.js:275:12)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Socket.stream.socket.on (internal/child_process.js:346:11)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at Pipe._handle.close [as _onclose] (net.js:557:12)
  killed: false,
  code: 127,
  signal: null,
  cmd: 'inkscape --without-gui \'build/.temp/boards/github.com/kitspace-forks/arduino-uno/unoptimized-top.svg\' --export-png=\'build/boards/github.com/kitspace-forks/arduino-uno/images/top-meta.png\' --export-height=400' }
[3/6] node -- tasks/page/processGerbers.js boards/github.com/kitspace-forks/Bus_Pirate boards...top-meta.png build/boards/github.com/kitspace-forks/Bus_Pirate/images/top-with-background.png
FAILED: build/boards/github.com/kitspace-forks/Bus_Pirate/images/top.svg 
[... cut ...]
    at Socket.emit (events.js:211:7)
    at Pipe._handle.close [as _onclose] (net.js:557:12)
  killed: false,
  code: 127,
  signal: null,
  cmd: 'inkscape --without-gui \'build/.temp/boards/github.com/kitspace-forks/Bus_Pirate/unoptimized-top.svg\' --export-png=\'build/boards/github.com/kitspace-forks/Bus_Pirate/images/top-meta.png\' --export-height=400' }
ninja: build stopped: subcommand failed.
BUILD FAILED
watching for file changes in src/ tasks/ boards/
./scripts/watch: line 26: inotifywait: command not found
Done in 16.24s.

kasbah, 2019-05-02:
If you look at the log you'll notice you are missing Inkscape. This needs to be added to the requirements in the readme actually.

evanyang1, 2019-05-02:
`npm i --save inkscape` Nice, but didn't work.
`sudo apt-get update`
`suod apt-get install inkscape` This works! It builds successfully!

kasbah, 2019-05-02:
You'll also need to install `inotifywait` via apt if you want to auto-rebuilding functionality.

evanyang1, 2019-05-02:
What else must be done before this issue can be closed?

kasbah, 2019-05-02:
Just what it says in the title.

commit 3797c67d (2020-02-16): Document Inkscape dependency 

fixes #173
closed 2020-02-16 by kasbah