kitspace/kitspace #183: Add ability to define readme path

kasbah, 2019-06-02:
Could just be a field in kitspace.yaml:

```yaml
readme: path/to/README.md
```

kasbah, 2019-10-29:
This is needed for re-defining the name in `multi` repos. E.g. for [this one](https://github.com/kitspace-forks/DIY_particle_detector/) ([preview](http://add-particle-detector.preview.kitspace.org/)). It's [currently failing](https://travis-ci.org/kitspace/kitspace/builds/604539267) because I can't redefine the names. 

Wondering if  the resident `multi` expert @SDBowen wants to look at his one?

SDBowen, 2019-10-29:
Yeah, I can work on this :+1:

kasbah, 2019-10-30:
I was thinking about it today. I guess this is an exercise we will have to repeat for every thing that involves a default path. Maybe it was a mistake to bind the `name` to the path for `multi`? Maybe we should just have the name and then a `path` field?

SDBowen, 2019-10-30:
I think adding a `path` field makes sense.

Wouldn't your example from above require both though, a `path` and `readme`? It seems there is no readme in the project folders?

kasbah, 2019-10-30:
Initially it's just adding `path` but eventually we should remove support for using the keys in `multi` as paths. 

No there are no readmes, but the reason it crashes when I change the name is that it's trying to `cd` into a non-existent directory, I assume to check if there is a readme (and maybe do other stuff that it doesn't need to do).

SDBowen, 2019-10-30:
Ah, I understand now.

So, we want to add a `path` field, but will still support the path in the `multi` key for the time being?

kasbah, 2019-10-30:
Yeah, just to let people transition. 

I think we still have a case for the readme field as well though so we should open a separate issue.