kitspace/kitspace #196: Generate OKH manifest for every project

kasbah, 2019-10-08:
We have standardised an [Open Know How Manifest](https://app.standardsrepo.com/MakerNetAlliance/OpenKnowHow/src/branch/master/1) to improve discoverability of open source hardware projects. We should generate these for every project.

julianstirling, 2020-02-26:
I think that now that https://okh.now.sh/ is online it would be a good idea to flesh it out with some kitspace boards. Especially if we can get the  manifests from field ready, so that we have a good balance of their stuff and kitspace stuff.

kasbah, 2020-02-27:
My plan had been to add it once there were at least an equal number (~90) projects with manifests so it doesn't flood the pool. 

I have had some further thoughts about this from the perspective of a "platform" though: People give their explicit consent to be added to kitspace.org but they never gave consent to be "associated" with OKH. 

Aisler.net had this problem. Back before they turned into a PCB service, they used to run a search site that crawled GitHub for electronics projects. They once told me that people complained to them that they never signed up to be listed on Aisler.net. 

Other platforms may have similar concerns and I am not sure what the best solution is. It may be that we need to gather approval from the project creators first or at least be very clear about the process of de-listing yourself. The equivalent for google is a robots.txt with:

```
User-agent: Googlebot
Disallow: /
```

julianstirling, 2020-03-07:
Yeah, I see the concern but aggregators exist if you put content on websites. Is there an easy way to notify people by email/through GitHub, saying that you will make manifests to make the projects easier to find and have it opt-in or opt-out in the kitspace yaml file?

kasbah, 2020-06-14:
I am currently leaning towards this solution: let people add an explicit field to their kitspace.yaml.

```yaml
okh: true
```

And we'll generate a manifest for these.

julianstirling, 2020-06-14:
Is there an easy way to contact people who are on kitspace to explain what okh is and ask them to consider enabling it? The trade-off there is between being too spammy, and no one setting it true.

kasbah, 2020-06-14:
Most projects are on Github, some on Gitlab. We can fairly easily send pull-requests to all of these.