kitspace/kitspace #199: Browser detection broken on Chrome

kasbah, 2019-10-22:
As noticed in #198 it seems Chrome browser can be detected as incompatible with 1-click BOM. That's an issue!

Browser detection happens here: https://github.com/kitspace/kitspace/blob/master/src/buy_parts/install_prompt.jsx#L32

kasbah, 2019-10-23:
I can't reproduce this with Chrome on Ubuntu. @SDBowen can you log what you get in a console on your chrome for `navigator.userAgent` and from `version` in the function above?

SDBowen, 2019-10-26:
``navigator.userAgent``:
"Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36"

``version``:
Chrome 77

It seems I only get this while running Kitspace locally. On the live site, I do not have an issue.

[image]

kasbah, 2019-11-08:
Ok yeah, this is actually as expected as we can't add localhost as a site the published extension can have access to.

closed 2019-11-08 by kasbah