kitspace/kitspace #221: Some links broken in tomu readme

kasbah, 2019-12-28:
For [tomu-hardware](https://kitspace.org/boards/github.com/im-tomu/tomu-hardware/), some links are broken in the readme. 

>Designs and documentation are available in:
>pcb/](./pcb): KiCad design files
>docs/](./docs): PDF schematics, pin-out diagrams, and 3D renders

@SDBowen maybe we can best solve this, and the `img` issue we are seeing in #220 by post-processing the readme html rather than pre-processing the markdown.

kasbah, 2020-01-02:
Also some issues with the new readme's for @ozel's project. 

http://add-particle-detector.preview.kitspace.org/boards/github.com/ozel/DIY_particle_detector/DIY%20Electron%20Detector/

https://raw.githubusercontent.com/ozel/DIY_particle_detector/master/hardware/V1.2/bom/kitspace_readme_electron-detector.md

SDBowen, 2020-01-03:
Will tackle this one next.

kasbah, 2020-01-06:
Maybe we could set up some testing infra for the readme processing so we avoid regressions for it in the future.

kasbah, 2020-01-11:
Had another thought: we may be able to pull in upstream fixes for link/image handling by upgrading marky-markdown. The reason I chose marky-markdown is that it's used by https://npmjs.com, and they aim for GitHub parity.

I did [fork](https://github.com/kasbah/marky-markdown/commit/ae8bbfa34b55c4a70d3238f4f3b87db5fecab21e) it because I feel GitHub and NPM over complicate the title-anchor handling by not just relying on the browser which would recognize a `name` attribute and scroll to it without needing additional JS. (Relevant [rejected pull-request](https://github.com/npm/marky-markdown/pull/232).) 

We would need to rebase that fork, though I am also not sure if there are any fixes in the newer marky-markdown that will fix our issues.

kasbah, 2020-02-16:
@SDBowen do you know when you are likely to have time for this? I'd like to fix the broken images but don't want double up efforts if you have already started.

SDBowen, 2020-02-16:
Yeah, sorry for not following through on this one. 

Going to take a look and see if I can solve this in a reasonable time. Will update you tomorrow.

ozel, 2020-02-17:
maybe helpful, a 'test case' on the kitspace master: 
https://kitspace.org/boards/github.com/ozel/diy_particle_detector/alpha-spectrometer/
generated from:
https://raw.githubusercontent.com/ozel/DIY_particle_detector/master/hardware/V1.2/bom/kitspace_readme_alpha-spectrometer.md

SDBowen, 2020-02-17:
I rebased your fork onto the current marky-markdown master, started up Kitspace and got a successful build, but an empty homepage :upside_down_face: I was hoping for an easy fix.

I have time this week to keep looking at it.

kasbah, 2020-02-17:
What does console say? Maybe jump on [chat](https://riot.im/app/#/room/#kitspace:matrix.org) to discuss debugging?

kasbah, 2020-02-22:
I am not having a lot of fun upgrading marky-markdown either. I think we should give up on that idea. I don't think it will solve our problems anyway since they are related to the processing we are doing ourselves.

SDBowen, 2020-02-24:
Sorry Kaspar, wasn't able to continue work on this. Going to take myself off this as I'm not sure when I will have the time... Will be back at some point.

I did hear about this [GitHub markdown API](https://developer.github.com/v3/markdown/) recently. Possible solution over marky-markdown or that is too many requests?

kasbah, 2020-02-24:
Ok, thanks for letting me know. Markdown API could work too, though I liked having a lib to do this. But marky-markdown is a pain in general. Might not get the `name` attributes from the API though (I really don't get why everyone uses JS for scrolling to a title, when browsers will do it when you put a `name` in there). Will think about it.

commit 6bee6f80 (2020-02-29): Fix image processing for blob to raw images

fixes #221
closed 2020-02-29 by kasbah