kitspace/kitspace #257: Mouser and LCSC bugs for ULX3S

emard, 2020-05-21:
HI

For our project ULX3S 
https://kitspace.org/boards/github.com/emard/ulx3s/

mouser doesn't exactly recognize part
Mouser: 700-MAX11125ATI+
and reports this (is ending with "+" making problems?)
More Than 1 Match Found for 700MAX11125ATI Select from a list of matching products

LCSC is very unreliable, I have to retry cca 10 times to get 48 parts added and 1 part
not added, but it's difficult to find which one and what is the problem about it.

kasbah, 2020-05-21:
Hi @emard, thanks very much for reporting. I will look into this.

emard, 2020-05-21:
Maybe I found 1, LCSC discontinued sticking it
but still uploading BOM is unreliable and I must retry many times

kasbah, 2020-05-21:
Sorry, could you put into other words? Not sure what you are saying at all. :smile:

kasbah, 2020-05-21:
Just tried it out with the ulx3s BOM loaded into 1-click BOM. It says 48 lines but when I add, only 46 are added but no errors are reported. This is what you are talking about, yeah?

<details title=BOM>
 <summary>BOM</summary>

```csv
References	Qty	Description	Manufacturer	MPN	Manufacturer	MPN	Manufacturer	MPN	Manufacturer	MPN	Digikey	Mouser	RS	Newark	Farnell	LCSC
AE1	1															
AUDIO1	1		CUI	SJ-43516-SMT-TR							CP-43516SJCT-ND	490SJ43515RSSMTTR				
B0, B1, B2, B3, B4, B5, B6	7			PTS645SL50SMTR92 LFS							CKN9088CT-ND	611PTS645SL50SMTR92				C285463
BAT1	1		Keystone	3000TR	MPD	BK-885	Harwin	S8201-46R			BK-885-ND	614SMTM1225.TR	1613707			C238097
C1, C3, C4, C5, C7, C8, C9, C11, C12, C15, C21, C54, C55, C58, C61	15			CC0805ZKY5V5BB226							1276-2413-1-ND	81GRM21BR61A226ME1L				C29277
C2, C6, C10, C18	4			CS0603KRX7R9BB471							399-9086-1-ND	710885012206081				C22396
C13, C16, C17, C19, C20, C22, C23, C24, C46, C47	10			CC0805MRX5R6BB225							587-5918-1-ND	80C0805C225K4R3123				C87995
C14, C36, C37, C38, C39, C40, C41, C42, C43, C44, C45, C59, C60, C62	14			CC0603KRX7R8BB224							1276-1112-1-ND	81GCM188R71E224K55D				C326805
C25, C26, C27, C28, C29, C30, C31, C32, C33, C34, C35, C48, C49, C50, C51, C52, C53	17			CC0603JRX7R8BB223							587-1238-1-ND	603CC603JRX7R8BB223				C107137
C56, C57	2			CC0603CRNPO9BN3R3							311-3848-1-ND	710885012006048				C85300
D0, D4, D11, D19	4			LTST-C170KRKT							160-1415-1-ND	743INS85ATR				C94868
D1, D5	2			LTST-C170KFKT							160-1413-1-ND	859LTSTC170KFKT				C284931
D2, D6	2			LTST-C170TGKT							160-1887-1-ND	859LTSTC170TGKT				C364562
D3, D7	2			LTST-C170TBKT							160-1579-1-ND	859LTSTC170TBKT				C375453
D8	1			STPS2L30AF							497-12310-1-ND	511STPS2L40AF				C111630
D9, D51, D52	3		Yageo	RC0805JR-070RL	ST	STPS2L30AF										
D10, D13, D14, D16, D17, D23, D24, D25, D26, D27, D29	11			1N4148WS							1N914BWSCT-ND	5121N914BWS				C258181
D12, D15	2		ON semiconductor	NSR0530HT1G	Diodes inc.	BAT54WSQ-7-F					NSR0530HT1GOSCT-ND	771BAT54J115				C12744
D18	1			LTST-C170TGKT							160-1887-1-ND	859LTSTC170TGKT				C364562
D20, D21	2			BZT52C3V6S							BZT52C3V6S-TPMSCT-ND	833BZT52C3V6STP				C353525
D22	1			LTST-C170TBKT							160-1579-1-ND	859LTSTC170TBKT				C375453
D28	1			MMDL914T1G												
GPDI1	1			10029449-111RLF		5-1903015-5					609-4614-1-ND	64910029449111RLF				C427307
H1, H2, H3, H4	4															
J1, J2	2															
J3	1															
J4	1															
J5	1															
L1, L2, L3	3		Murata	DFE322512F-2R2M=P2							490-18405-1-ND	81DFE322520FD2R2MP2				C285914
L4, L5	2			CV201210-330K							CV201210-330KCT-ND	652CV201210330K				C412010
LCD1	1			DM-OLEDC095-626												
Q1	1			BC857							TBC857BLMCT-ND	583BC856AT				C181146
Q2	1			2N7002P,215							2N7002K-TPMSCT-ND	757T2N7002AKLM				C179727
R1, R3, R4, R25, R26, R30, RB2	7			AC0603FR-104K7L							YAG3613CT-ND	755SFR03EZPF4701				C218056
R2, RB1	2			RC0603FR-1018KL							311-18.0KHRCT-ND	71CRCW060318K0FKEAC				C425614
R5	1			RC0603FR-132M2L							P2.20MHCT-ND	603AC0603FR072M2L				C18679
R6, R8, R14, R18, R29, R40, R54, R57, R66	9			RC0603FR-071K1L							P1.10KHCT-ND	603RC0603FR071K1L				C269415
R7, R10, R17, R21, R39, R51, R60	7			RC0603FR-07130RL							311-130HRCT-ND	603RC0603FR07130RL				C269457
R9, R13, R31, R32, R33, R34, R63, R64, RA1, RA2, RA3	11			RC0603FR-0715KL							YAG3569CT-ND	603RC0603FR0715KL				C217750
R11, R12, R27, R28	4			RC0603FR-0710KL							311-10KLMCT-ND	603RC0603FR0710KL				C15401
R15, R19, R35, R36, R37, R41, R42, R43, R44, R45, R46, R47, R48, R58, R61, R62, R65	17			RC0603FR-07549RL							311-549HRCT-ND	603RC0603FR07549RL				C308281
R16, R20, R59	3			RC0603FR-07270RL							311-270HRCT-ND	603RC0603FR07270RL				C325609
R22, R23, RB3	3			RT0603FRE073K3L							YAG3605CT-ND	603RC0603FR073K3L				C425629
R24	1			AC0603FR-10100KL							P100KHCT-ND	71RCA0603100KJNEC				C73806
R38	1		CTS	73L2R47J							73L2R47JCT-ND	77473L2R47J				
R49, R50, R52, R53	4			AC0603JR-0727RL							311-27.0HRCT-ND	603AC0603JR0727RL				C325726
R56	1			RC0603JR-070RL												
RP1, RP2, RP3	3			RC0805JR-070RL												
RV2	1			RC0805JR-070RL												
RV3	1			RC0805JR-070RL							YAG1327CT-ND	71RCG08050000Z0EA				C112378
SD1	1			SCHD3A0100		47219-2001					WM6698DKR-ND	538472192001				C164170
SW1	1			219-4MST		DSHP04TS-S					CT2194MST-ND	7742194MST				C319050
U1	1		Lattice	LFE5U-85F-6BG381C							220-2071-ND	842LFE5U85F6BG381C				
U2	1		Micron	MT48LC16M16A2TG-6A	ISSI	IS42S16160G-7TL-TR	Winbond	AS4C32M16SB-7TCN	Alliance	M12L2561616A-6TG2T	1450-1385-1-ND	913AS4C32M16SB7TCN				C492195
U3, U4, U5	3		TI	TLV62569DBVR	DIOO	DIO6015BCST5					296-47360-1-ND	595TLV62569DBVR				C141836
U6	1			FT231XS							768-1129-1-ND	895FT231XSR	7570032			C132160
U7	1		https://www.nxp.com/docs/en/packing/SOT96-1_118.pdf	MCP7940NT-I/SN		PCF8523T/1,118					MCP7940NT-I/SNCT-ND	579MCP7940NTI/SN				C148052
U8	1			MAX11125ATI+		MAX11123ATI+					MAX11125ATI+-ND	700MAX11125ATI+				C490299
U9	1			ESP-WROOM-32							1904-1010-1-ND	356ESPWROOM3216MB				C529581
U10	1			IS25LP128F-JBLE		IS25LP032D-JNLE-TR		IS25LP032D-JNLA3		W25Q128JVSIM	W25Q128JVSIQCT-ND	454W25Q128JVSIMTR				C113767
U11	1			PCA9306D							568-4215-1-ND	771PCA9306DT		70R6428		
US1, US2	2		FCI	10118192-0001LF	Molex	47346-0001					WM17141CT-ND	649101181920001LF				C132564
Y1	1			FNETHE025		SiT8008BC-81-33E-25.000000X		S7D25.000000D20F30T			FNETHE025CT-ND	729FNETHE025				C390569
Y2	1			LFXTAL016178REEL		CM7V-T1A-32.768k-9pF-20PPM-TA-QA T3		LFXTAL003000Reel			1923-1021-1-ND	449LFXTAL016178REEL				C252402
```

</details>

emard, 2020-05-21:
yes yes that is it,
sometimes it says 48 failed from 48, sometimes other number of lines failed
and sometimes it all passes except 1, that is the best I can do in let's say 10
repeated attempts.

Maybe this 1 I can "fix" by removing a part that I think is making problems
because it is no longer stocked on LCSC but that's just a guess because
I don't know which 1 failed

emard, 2020-05-21:
I tried again, it's random, this is how it "works" for me
(clicking LCSC, scrolling at items for few seconds,
 emptying cart if there are any items in the cart and
repeating click LCSC again...)

48 of 48 lines failed
6 of 48 lines failed
16 of 48 lines failed
2 of 48 lines failed
48 of 48 lines failed
48 of 48 lines failed
48 of 48 lines failed
48 of 48 lines failed