kitspace/kitspace #273: Improve search

kasbah, 2020-06-10:
Search will need to be improved as the volume of projects increases. Just dropping some relevant links here for now:

- https://lunrjs.com/
- https://github.com/typesense/typesense
- https://stork-search.net/
- https://endler.dev/2019/tinysearch/ ([on HN](https://news.ycombinator.com/item?id=23473303))
- https://github.com/nextapps-de/flexsearch

kasbah, 2020-06-15:
I should also mention here that we are [working on integrating gitea](https://github.com/kitspace/kitspace-using-gitea) as a back end which already has a server-side built in search we should re-purpose if it's good enough.

jasonbosco, 2020-10-25:
👋 I work on Typesense and just stumbled on this. Please let me know if I can help as you explore Typesense.

kasbah, 2020-10-25:
Thanks for the offer @jasonbosco. Do you think Typesense could be used to augement the search when we move to [Gitea](https://github.com/go-gitea/gitea/) as a backend?

jasonbosco, 2020-10-26:
Hi @kasbah, I don't see why not. But could you expand on your use of Gitea for a backend? Are you just looking to host this git repo in Gitea, or are you looking to use Gitea as your data store?

In any case, as long as you can extract the data as structured JSON data, you can index it to Typesense and use it to power the search experience.