kitspace/kitspace #279: Take BOM into account when searching

kasbah, 2020-06-15:
For instance [this project](https://kitspace.org/boards/github.com/dusjagr/pn001_pop-neuron_shenzhenready/) uses a TL072ACP as it's main chip. But when I search for "TL072ACP" I get nothing. We should fix this.

kasbah, 2020-06-16:
Copying here from https://riot.im/app/#/room/#kitspace:matrix.org/$159234042467995mArvq:matrix.org. Thanks @AbdulrhmnGhanem for a detailed run through of the issues you are seeing. 

> - scenario 1: Just modify correctTypes to include the BOM. Then adding the BOM to the filters.
>   - The BOM is partially ready; just find the files name info.json in the build/.temp/boards folder recursively. In this case I have no idea which part of the site gets built first the, is it safe to assume the ProcessBOM will run before makeBoardsJson?

I am not sure this will help you, but resolving the graph of what needs to be built before what is what ninja and the ./configure script is for. 

>   - Duplicate the code for parsing the BOM from ProcessBOM into makeBoardJson.

I would be ok with this. Maybe we can just move it?

>   -Either way, I think you better do it yourself:
>   You have deeper understanding of the build process.

If I had found the time, I would have done it ;) 

>   - I'm not OK with adding another level of complexity to already complicated code base.
> - Senario 2: Integrate typesense, in this case I'm happy to isolate the search feature on a docker container by itself.

I am not sure this will help. We'd better put it off till we integrate Gitea in that case I think.

> Why didn't you use any static site generator instead of building your own as config?

I built this site before NextJS or Gatsby etc existed. Webpack wasn't around or not popular yet. We also have a lot of custom processing of assets that these build systems and static site generators don't give use. Anyway, in the kitspace-using-gitea version we are using NextJS.

AbdulrhmnGhanem, 2020-06-16:
> I am not sure this will help you, but resolving the graph of what needs to be built before what is what ninja and the ./configure script is for.

Oh, I told you I'm not familiar with ninja, I'll give this a shot.

kasbah, 2020-06-16:
It's a cool build system, worth learning. It is used to build the Chrome browser for instance. Low-level so it lets you make your own build system more easily.

In a way it's like `make` but much less complicated. It's made for making another script, like our`./configure` to generate the makefile equivalent: the `build.ninja` file.

AbdulrhmnGhanem, 2020-06-18:
It started to make sense. The problem I'm currently facing is to define the dependencies(`info.json` files for all boards), another dependency is needed (`boards.json`).
```javascript
const fs = require('fs');

if (require.main !== module) {
    module.exports = function (config) {
        const targets = ['build/.temp/search_index.json'];

        const boardsPath = 'build/.temp/boards.json';
        const boardsInfo = JSON.parse(fs.readFileSync(boardsPath).toString());
        const boardsInfoPaths = boardsInfo
            .map(b => `build/.temp/boards/${b.id}/info.json`);

        const deps = [boardsPath, ...boardsInfoPaths];

        return { deps, targets, moduleDep: false };
    }
} else {
    // indexing logic here
}
```
Otherwise, all the boards will have to be indexed individually and it's not a matter of parsing the same YAML files twice, it's more time consuming than that; the 1-click-bom will run twice, and it's slow. I tried not to guess and profile the build process with [buildbloat](https://github.com/nico/buildbloat) but the script assumes a standard format for `build.ninja` and it does not match kitspace's build log but the fact I can see it in terminal (the 1-click-bom task) means it's already slow and doubling it wouldn't be a wise move.
So is there a way to define this kind of dependencies?

kasbah, 2020-06-18:
Great it's making some sense at least. I am not sure I totally understand the issue though. Would it help if we just keep `boards.json` as is and make another `search_index.json` or something that is built from `boards.json` and all the project `info.json` files?

AbdulrhmnGhanem, 2020-06-18:
I did exactly this, you will find it as the `target` in the snippet above. The problem is to specify all `info.json` as a dependencies `boards.json` must be read first. The dependencies can only be generated after `boards.josn` is written to the desk. The snippet above fails with `the file(boards.json) does not exist`.
I got an idea while writing the comment, I'll try it.

kasbah, 2020-06-18:
Oh sorry, I missed that! If you pull-request your code it's a little easier to review.

AbdulrhmnGhanem, 2020-06-18:
#285 the dependencies worked out fine on the dev config, but it doesn't resolve properly on Travis.

kasbah, 2020-06-19:
Why are you closing this? Close this when the branch get's merged.

PR #286 (2020-06-23): Use bom for search
Create a new search index for boards, taking BOM into account. More details at #285. 
This should close #279.
closed 2020-06-23 by kasbah