kitspace/kitspace #302: Links can be wrong

kasbah, 2020-09-06:
A Twitter user reports that the links on the front page can be wrong sometimes. This disappeared when loading is hard to reproduce though. Here are some screenshots. 

[image]

[image]

[image]

kasbah, 2020-09-06:
Chromium version: 84.0.4147.105 (Official Build) Built on Ubuntu, running on Ubuntu 18.04 (64-bit)

6r1d, 2020-09-06:
Hey, I did the screensots, I should've noticed the GitHub link. :-)
I am not sure, but I think links are broken due to loading many pages.