kitspace/kitspace #309: ulx3s gerbers not yet pulled

emard, 2020-11-04:
HI

I made last commit few days ago and at ulx3s gerbers should be version v3.1.4
but on kitspace it is still v3.1.3
Do I just have to wait or should change something from my side

kasbah, 2020-11-04:
Hi emard, thanks for pointing it out. 

We had some problems with the build, which we worked around by disabled caching. Now https://gogs.zenerdio.de/sebastian/satnogs-diplexer went down. Since we were running uncached we didn't have a copy of it to use instead so the update just fails for everyone instead of removing that page.

Anyway, bear with me, I'll fix it by carefully running a cached build.

kasbah, 2020-11-05:
It's been updated.

closed 2020-11-05 by kasbah

emard, 2021-05-24:
HI!

on ulx3s something is wrong with copper layers, somehow

On inspect gerbers, it detects 2 top and 2 bottom copper layers,
first layer is empty other one is normal. Inner layers are ok.

On the board view on main page, there is no yellow copper color.

kasbah, 2021-05-24:
Hey, thanks for reporting. I believe this is due to the Tracespace library update and https://github.com/tracespace/tracespace/issues/357. If you set "View Filenames" on the Tracespace viewer you can see it's matching the PDF files:

[image]

I will try and fix this issue in Tracespace at some point. Until then the work-around is to not put the PDF files in the same folder as the Gerber files. I would suggest making a `plot/gerbers` folder, putting all the Gerbers in there and changing the path in the kitspace.yaml.

emard, 2021-05-24:
thanx for the hint, I made separate directory for the gerbers without PDF
let's see how will this improve the rendering!