kitspace/kitspace #311: out-of-stock handler

emard, 2020-11-18:
HI

While maintainig BOM I often have to fix parts that are currently out of stock,
usually I keep 2-3 alternatives and almost always something from there is on stock.

Is there some automatic 1-click-bom feature which will auto-resolove out-of-stock 
if I specify multiple part number equivalents in fields like Mouser1, Mouser2, ...

kasbah, 2020-11-19:
Hi, the closest thing we have to this currently is the bom-builder. There you can't specify alternatives quite like you describe it but it will find alternatives and in-stock components for you. Do you have access to the bom-builder demo/beta?

emard, 2020-11-19:
HI

I didn't know it exists but it looks like useful tool.

Still I'd like to have fixed set of alternatives because
not all similar function pin compatible components really
fit (like for example choice of power inductors) but it needs
a human to compare datasheets and prepare a set of valid
equivalents.

kasbah, 2020-11-19:
It's not exactly like you describe but it does let you select different distributor parts depending on stock. Pin-compatibility doesn't come into it, only the MPNs and description you provide. If you want to try it, email me at kaspar@kitspace.org and I'll give you access. It would probably help if we have a short session on Zoom or similar so I can explain it a little.

Anyway, I'll leave this issue open as a reminder to support specifying alternatives in the BOM file itself.