kitspace/kitspace #331: [FEATURE REQUEST] Add ability to support project owner to project page

darigovresearch, 2021-04-24:
Following discussion [here](https://github.com/MaslowCNC/Electronics/issues/57) thought it would be good to have a way to support the project owner of the kitspace project directly on the page.

A first iteration could be a field in the .yml file that would render a link to a Patreon/donation page etc but this could morph into a slider for example that takes a percentage of the cost of the BOM from the supplier and applies an optional selector to support both Kitspace and the project creator to help fund development/maintenance costs.

Example of a good implementation of this [Humble bundle](http://humblebundle.com/) when buying book bundles to split cost of purchase between Humble, the supporting charity & the book publisher making it sustainable for each party involved.

Any thoughts?

kasbah, 2021-04-24:
Not a bad idea. What we are planning to implement on v2 are group-buys where the creator would get a cut. Here's a mock-up of how that could look: https://balsamiq.cloud/ssnon0j/paket19/rF42B

kasbah, 2021-04-24:
Oh and what I wanted to say: a quick way to do it right now is just to add a link or a badge to the README.md, that will appear on Kitspace too.

darigovresearch, 2021-04-25:
Group buys are a very intriguing feature specifically & having the support creator on that feature would be beneficial as well. The badge work around is a good one if the project owner knows how to use badges, potentially worth mentioning in the documentation if possible in the interim.

The support feature may not need to wait till the V2 depending on the demand for it, potential rollout could be as follows if anyone from the community wishes to implement

- Badges (current interim solution to be documented if possible)
- Add support section to .yml which renders at the top of the page
- Add selector price support 
- V2 group buy with tip to project owner and/or kitspace