kitspace/kitspace #332: [FEATURE REQUEST] Add to cart feature to shop

darigovresearch, 2021-04-24:
Can see that there will be people who will want to pay for multiple items in the shop in one go and to calculate shipping together to avoid having to pay twice for shipping.

Is this possible?

kasbah, 2021-04-24:
No, not currently. The shop part will be removed again I think, once I get tired of making particle detector kits. I should really remove the ruler project from the shop now. People don't really order it and the latest batch has a small mistake in it, I don't want to sell it like that. If you want to order some particle detectors I can just throw in the rulers for free :)

darigovresearch, 2021-04-24:
You never know a new project may come along that may peak you interest in producing kits again in the future. Having kits are beneficial (mainly because there is still some uncertainty when ordering components & PCBs that they may not be correct) so hope it sticks around or if project holders can host their own on the site but full fill it themselves. Also if the website is open source having cart functionality may be an easy feature to build by the community and may help other users as well.

There are many PCB rulers that can be found online with references for component/trace width/through hole sizes to help when doing CAD but we liked this one as you could actually populate it with components and it does something.

Out of curiosity what was the small mistake in case we make the same error in the future?

kasbah, 2021-04-25:
Well, it's not really on our roadmap to continue the shop for kits or allow people to fulfill themselves. If we do allow people to fulfill themselves I am not even sure it would be part of the same code-base of the current shop. So, I am not sure keeping this issue open makes sense as it would be hard to figure out how to work on it. As mentioned in another issue, we are working on a group-buy feature on v2 and having a cart there would make more sense. 

The error on the ruler is with the font, I must have not had the font when I re-exported. You can see it on the words "svg2shenzen" and "v-score" on the back. The words don't fit in the space anymore.  https://tracespace.io/view/?boardUrl=https://kitspace.org/boards/github.com/kitspace/ruler/ruler-2af1eef-gerbers.zip

closed 2021-04-25 by kasbah