kitspace/kitspace #404: Multiple boards not being picked up?

amchagas, 2022-12-01:
Hi folks!
I am probably doing something silly on my side, but we recently updated our project, which was already configured to display multiple boards on kitspace (and working properly). 

We had a bit of an update to our repository and would like to include more boards that were created in the meantime... So we updated the [kitspace.yaml](https://github.com/BeeHive-org/BeeHive/blob/master/kitspace.yaml) file some days ago, but this does not seem to have translated into more boards on kitspace.org. I wonder if we need to resubmit a pull request, or if there is something we are missing?

Thanks!

kasbah, 2022-12-01:
Hey, [it's complaining](https://github.com/kitspace/kitspace/actions/runs/3594132741/jobs/6052000691#step:4:467) that `hardware/PCBs/grove_adaptor/1-click-bom.csv` doesn't exist.

By the way I would recommend to removing the `board/` prefix in the names. We will stop supporting the `/` in names at some point. I'll add some redirects for your existing projects.

isobianin, 2022-12-02:
Hi, thanks for the explanation, added respective .csv files and I think [this commit] (https://github.com/BeeHive-org/BeeHive/commit/92f42316d5c8ca38608c763e6b6f1eb82a24310f) should tackle the issue

kasbah, 2022-12-02:
Looks like they are all being picked up now? I see one small rendering problem with the panel (there are likely some gaps in the board outline) but otherwise it looks great.

[image]

isobianin, 2022-12-02:
Yeah, there are :sweat_smile:. Thanks for looking into this!

kasbah, 2022-12-02:
It's not your fault. Ideally our renderer would be better at finding the right shape when small gaps like this are present.

closed 2022-12-02 by kasbah