kitspace/kitspace #416: Outdated PCB design shown

NuclearPhoenixx, 2023-11-12:
Hi there,

Regarding my project:
- Repo: https://github.com/OpenGammaProject/MicroFC-SiPM-Carrier-Board
- Kitspace: https://kitspace.org/boards/github.com/opengammaproject/microfc-sipm-carrier-board/

The PCB currently shown on Kitspace is outdated unfortunately. As you can see in the files in the repo, the current release is 2.0, while Kitspace still has the 1.3 revision. Not sure if it's something with my Kitspace config or if this is an issue on your end. This looks to be a problem with the entire GitHub repo since not only the Gerber file and the BOM are wrong, but this also extends to the README as well.

kasbah, 2023-11-12:
Hey, sorry about this, it's an issue on our end. I will fix it as soon as I can.

kasbah, 2023-11-13:
Does this look right? https://review.staging.kitspace.dev/OpenGammaProject/MicroFC-SiPM-Carrier-Board (Not really a public site, just our staging for kitspace-v2)

NuclearPhoenixx, 2023-11-13:
Gerber and README look good, but the BOM is still the old one.

kasbah, 2023-11-13:
Ok, thanks, now it's good?

NuclearPhoenixx, 2023-11-30:
Hey, uh I'm not sure if this _should_ be resolved by now, but I still see the old files.

How often do you query the repos for files? Cause all the repos below also don't show the latest files: 😕 

https://kitspace.org/boards/github.com/opengammaproject/mini-sid/
https://kitspace.org/boards/github.com/opengammaproject/microfc-sipm-array-board/
https://github.com/OpenGammaProject/Open-Gamma-Detector

Are you trying to address this with kitspace-v2? If so, do you have an ETA on that? :)

NuclearPhoenixx, 2023-12-01:
Okay, well then I'll just wait for v2. Thanks for clarifying!